linux-usb-gadgets/libusbgx #5: Isochronous endpoint support?

bertrik, 2017-02-24:
Is is possible to use this library and/or configfs for creation of an USB device with an isochronous endpoint? Or can I only create devices with functions that are pre-defined in the usbg_function_type enum?

kopasiak, 2017-02-28:
I'm not sure what you really want to do but:

If you would like to write your own function you should use FunctionFS because this library is only for composing USB device not for implementing own Function.

If you would like to use some kernel funciton which uses iso endpoints then you need to add its support  to libusbgx or use ConfigFS directly

bertrik, 2017-03-06:
I see now that it's not really the point of libusbg to implement functions.
I'll close the issue, thanks for your reply.

closed 2017-03-06 by bertrik