linux-usb-gadgets/libusbgx #12: usbg_write_buf() can fail without reporting it

notro, 2017-11-18:
I had a problem where usbg_enable_gadget() reported success, but it should have failed:
```
[869503.937381] configfs-gadget 20980000.usb: failed to start g1: -19
```

I haven't got time to make a PR, but the following worked for me, now returning USBG_ERROR_NO_DEV.

```diff
diff --git a/src/usbg_common.c b/src/usbg_common.c
index de9e40d..6321993 100644
--- a/src/usbg_common.c
+++ b/src/usbg_common.c
@@ -172,7 +172,9 @@ int usbg_write_buf(const char *path, const char *name,
                        ret = USBG_ERROR_IO;
        }

-       fclose(fp);
+       if (fclose(fp) == -1 && ret > 0) /* don't mask previous error */
+               ret = usbg_translate_error(errno);
+
 out:
        return ret;
 }
@@ -192,7 +194,7 @@ int usbg_write_int(const char *path, const char *name, const char *file,
        if (ret > 0)
                ret = 0;

-       return 0;
+       return ret;
 }

 int usbg_write_string(const char *path, const char *name,
@@ -204,7 +206,7 @@ int usbg_write_string(const char *path, const char *name,
        if (ret > 0)
                ret = 0;

-       return 0;
+       return ret;
 }

 int ubsg_rm_file(const char *path, const char *name)
```

kopasiak, 2017-11-20:
Could you please check if #13 fixes the issue for you?

notro, 2017-11-20:
Yes it worked, thanks.

kopasiak, 2017-11-20:
ok. merged. Thank you for reporting this issue:)

closed 2017-11-20 by kopasiak