linux-usb-gadgets/libusbgx #14: usbg_disable_gadget() doesn't work

notro, 2017-11-23:
I get USBG_ERROR_BUSY when using usbg_disable_gadget().

The problem is an off-by-one error in the kernel:
```diff
diff --git a/drivers/usb/gadget/configfs.c b/drivers/usb/gadget/configfs.c
index 502a096fc380..fbc3a9444805 100644
--- a/drivers/usb/gadget/configfs.c
+++ b/drivers/usb/gadget/configfs.c
@@ -260,8 +260,8 @@ static ssize_t gadget_dev_desc_UDC_store(struct config_item *item,
        name = kstrdup(page, GFP_KERNEL);
        if (!name)
                return -ENOMEM;
-       if (name[len - 1] == '\n')
-               name[len - 1] = '\0';
+       if (name[len - 2] == '\n')
+               name[len - 2] = '\0';

        mutex_lock(&gi->lock);
```

This also works:
```diff
diff --git a/src/usbg.c b/src/usbg.c
index 3a2c674..fc74167 100644
--- a/src/usbg.c
+++ b/src/usbg.c
@@ -2232,7 +2232,7 @@ int usbg_disable_gadget(usbg_gadget *g)
        if (!g)
                return ret;

-       ret = usbg_write_string(g->path, g->name, "UDC", "\n");
+       ret = usbg_write_string(g->path, g->name, "UDC", "");
        if (ret != USBG_SUCCESS)
                goto out;
```

kopasiak, 2017-11-24:
Thank you for this bug report. I see now where the problem is but I think it should be fixed in a little bit different way. I'll prepare  a patch for both libusbgx and kernel in next week.

falstaff84, 2017-12-07:
Hit this issue too today. Strace shows that libusbgx is also sending the 0 byte:

    open("/sys/kernel/config/usb_gadget/grndis/UDC", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 27
    fstat64(27, {st_mode=S_IFREG|0644, st_size=0, ...}) = 0
    write(27, "\n\0", 2)                    = -1 EBUSY (Device or resource busy)
    close(27)                               = 0

I don't think libusbgx should do that. It's just a off by one error in usbg_write_string.

PR #18 (2017-12-11): Fix usbg_write_string off by one error
This fixes #14.
closed 2017-12-11 by kopasiak