linux-usb-gadgets/libusbgx #19: Empty instance name does not allow re-initialization

falstaff84, 2017-12-09:
Creating a mass storage function using an empty instance name e.g. by calling `usbg_create_function(g_ms, USBG_F_MASS_STORAGE, "", &f_ms_attrs, &f_ms)` seems to work fine and allow to use the gadget device.

However, trying then to reinitialize the configuration using `usbg_ret = usbg_init("/sys/kernel/config", &s);` fails with -3 `USBG_ERROR_INVALID_PARAM`.

One obvious fix is to use a non-empty instance name.

kopasiak, 2017-12-11:
Hmmm... Generally empty instance name should illegal from kernel point of view at least because of functionfs (it's quite hard to mount a device with "" name). So  I'll send kernel patch to fix this and if upstream accepts it I'll also disallow this in libusbgx

falstaff84, 2017-12-12:
I see, seems reasonable.

PR #20 (2018-01-15): Disallow creating function with empty instance name
USB function should have both type and instance name.
Mistakenly it was possible to create function with empty instance
name. This may lead to situations like FunctionFS device name
identified by "" or some misleading debug messages from TCM like:

tcm: Activating

Additionaly trying to execute usbg_init() after creating such
a function fails with -3 USBG_ERROR_INVALID_PARAM because it checks
if instance name is not empty.

To avoid this let's add a check in usbg_create_function() that
instance name should have at least one character.

This f [... cut]
closed 2018-01-15 by kopasiak