linux-usb-gadgets/libusbgx #21: Failing to build documentation

battlesnake, 2018-01-09:
In a fresh clone, I've run:

```
autoreconf -i
./configure
make
make doxygen-doc
```

The error log is at the end of this post since it's quite long.  I'm not too familiar with autotools or Doxygen, so let me know if there's any extra info that would be useful for diagnosing this.

Tools/packages in use:

```
$ autoreconf --version
autoreconf (GNU Autoconf) 2.69
...

$ make --version
GNU Make 4.2.1
...

$ doxygen --version
1.8.13

$ pdflatex --version
pdfTeX 3.14159265-2.6-1.40.18 (TeX Live 2017/Arch Linux)
kpathsea version 6.2.3

$ pacman -Qo `which pdflatex`
/usr/bin/pdflatex is owned by texlive-bin 2017.44590-9

$ pacman -Qi texlive-bin
Name            : texlive-bin
Version         : 2017.44590-9

```

Output log:

```
This is pdfTeX, Version 3.14159265-2.6-1.40.18 (TeX Live 2017/Arch Linux) (preloaded format=pdflatex)
 restricted \write18 enabled.
entering extended mode
(./refman.tex
LaTeX2e <2017-04-15>
Babel <3.15> and hyphenation patterns for 84 language(s) loaded.
(/usr/share/texmf-dist/tex/latex/base/book.cls
Document Class: book 2014/09/29 v1.4h Standard LaTeX document class
(/usr/share/texmf-dist/tex/latex/base/bk10.clo))
(/usr/share/texmf-dist/tex/latex/base/fixltx2e.sty

Package fixltx2e Warning: fixltx2e is not required with releases after 2015
(fixltx2e)                All fixes are now in the LaTeX kernel.
(fixltx2e)                See the latexrelease package for details.

) (/usr/share/texmf-dist/tex/latex/tools/calc.sty) (./doxygen.sty
(/usr/share/texmf-dist/tex/latex/base/alltt.sty)
(/usr/share/texmf-dist/tex/latex/tools/array.sty)
(/usr/share/texmf-dist/tex/latex/float/float.sty)
(/usr/share/texmf-dist/tex/latex/base/ifthen.sty)
(/usr/share/texmf-dist/tex/latex/tools/verbatim.sty)
(/usr/share/texmf-dist/tex/latex/xcolor/xcolor.sty
(/usr/share/texmf-dist/tex/latex/graphics-cfg/color.cfg)
(/usr/share/texmf-dist/tex/latex/graphics-def/pdftex.def)
(/usr/share/texmf-dist/tex/latex/colortbl/colortbl.sty))
(/usr/share/texmf-dist/tex/latex/tools/longtable.sty)
(/usr/share/texmf-dist/tex/latex/tabu/tabu.sty
(/usr/share/texmf-dist/tex/latex/varwidth/varwidth.sty))
(/usr/share/texmf-dist/tex/latex/tools/tabularx.sty)
(/usr/share/texmf-dist/tex/latex/multirow/multirow.sty))
(/usr/share/texmf-dist/tex/latex/adjustbox/adjustbox.sty
(/usr/share/texmf-dist/tex/latex/xkeyval/xkeyval.sty
(/usr/share/texmf-dist/tex/generic/xkeyval/xkeyval.tex
(/usr/share/texmf-dist/tex/generic/xkeyval/xkvutils.tex
(/usr/share/texmf-dist/tex/generic/xkeyval/keyval.tex))))
(/usr/share/texmf-dist/tex/latex/adjustbox/adjcalc.sty)
(/usr/share/texmf-dist/tex/latex/adjustbox/trimclip.sty
(/usr/share/texmf-dist/tex/latex/graphics/graphicx.sty
(/usr/share/texmf-dist/tex/latex/graphics/graphics.sty
(/usr/share/texmf-dist/tex/latex/graphics/trig.sty)
(/usr/share/texmf-dist/tex/latex/graphics-cfg/graphics.cfg)))
(/usr/share/texmf-dist/tex/latex/collectbox/collectbox.sty)
(/usr/share/texmf-dist/tex/latex/adjustbox/tc-pdftex.def))
(/usr/share/texmf-dist/tex/latex/ifoddpage/ifoddpage.sty))
(/usr/share/texmf-dist/tex/latex/base/inputenc.sty
(/usr/share/texmf-dist/tex/latex/base/utf8.def
(/usr/share/texmf-dist/tex/latex/base/t1enc.dfu)
(/usr/share/texmf-dist/tex/latex/base/ot1enc.dfu)
(/usr/share/texmf-dist/tex/latex/base/omsenc.dfu)))
(/usr/share/texmf-dist/tex/latex/base/makeidx.sty)
(/usr/share/texmf-dist/tex/latex/tools/multicol.sty)
(/usr/share/texmf-dist/tex/latex/base/textcomp.sty
(/usr/share/texmf-dist/tex/latex/base/ts1enc.def
(/usr/share/texmf-dist/tex/latex/base/ts1enc.dfu)))
(/usr/share/texmf-dist/tex/latex/wasysym/wasysym.sty)
(/usr/share/texmf-dist/tex/latex/base/fontenc.sty
(/usr/share/texmf-dist/tex/latex/base/t1enc.def))
(/usr/share/texmf-dist/tex/latex/psnfss/helvet.sty)
(/usr/share/texmf-dist/tex/latex/psnfss/courier.sty)
(/usr/share/texmf-dist/tex/latex/amsfonts/amssymb.sty
(/usr/share/texmf-dist/tex/latex/amsfonts/amsfonts.sty))
(/usr/share/texmf-dist/tex/latex/sectsty/sectsty.sty)
(/usr/share/texmf-dist/tex/latex/geometry/geometry.sty
(/usr/share/texmf-dist/tex/generic/oberdiek/ifpdf.sty)
(/usr/share/texmf-dist/tex/generic/oberdiek/ifvtex.sty)
(/usr/share/texmf-dist/tex/generic/ifxetex/ifxetex.sty))
(/usr/share/texmf-dist/tex/latex/fancyhdr/fancyhdr.sty)
(/usr/share/texmf-dist/tex/latex/natbib/natbib.sty)
(/usr/share/texmf-dist/tex/latex/tocloft/tocloft.sty)
Writing index file refman.idx
(/usr/share/texmf-dist/tex/latex/hyperref/hyperref.sty
(/usr/share/texmf-dist/tex/generic/oberdiek/hobsub-hyperref.sty
(/usr/share/texmf-dist/tex/generic/oberdiek/hobsub-generic.sty))
(/usr/share/texmf-dist/tex/latex/oberdiek/auxhook.sty)
(/usr/share/texmf-dist/tex/latex/oberdiek/kvoptions.sty)
(/usr/share/texmf-dist/tex/latex/hyperref/pd1enc.def)
(/usr/share/texmf-dist/tex/latex/latexconfig/hyperref.cfg)
(/usr/share/texmf-dist/tex/latex/hyperref/backref.sty
(/usr/share/texmf-dist/tex/latex/oberdiek/rerunfilecheck.sty))
(/usr/share/texmf-dist/tex/latex/url/url.sty))

Package hyperref Message: Driver: hpdftex.

(/usr/share/texmf-dist/tex/latex/hyperref/hpdftex.def)
(/usr/share/texmf-dist/tex/latex/hyperref/puenc.def)
(/usr/share/texmf-dist/tex/latex/caption/caption.sty
(/usr/share/texmf-dist/tex/latex/caption/caption3.sty)
(/usr/share/texmf-dist/tex/latex/caption/ltcaption.sty)) (./refman.aux

LaTeX Warning: Label `group__libusbgx_gadfb1a7e1855f9d8072f57d6d51ce0402' multi
ply defined.

LaTeX Warning: Label `group__libusbgx_ga49d7ddf93131e92ac5d8f0a24ad96f56' multi
ply defined.

LaTeX Warning: Label `group__libusbgx_ga61197026a63c0f9bc95f2c1d39308c59' multi
ply defined.

LaTeX Warning: Label `group__libusbgx_ga0cd65b6b5cfa50387cc9b0ea8bbcc832' multi
ply defined.

LaTeX Warning: Label `group__libusbgx_ga53ff55df4793b365cb6bf2723053d018' multi
ply defined.

! LaTeX Error: Missing \begin{document}.

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...

l.884 e
       r}{Index}{107}{section*.46}}
?
! Emergency stop.
 ...

l.884 e
       r}{Index}{107}{section*.46}}
!  ==> Fatal error occurred, no output PDF file produced!
Transcript written on refman.log.
mv: cannot stat 'refman.pdf': No such file or directory
make: *** [Makefile:1018: doxygen-doc/libusbgx.pdf] Error 1
make: *** Waiting for unfinished jobs....
```

kopasiak, 2018-01-10:
I've just check the doc build on my machine and it works fine.
My setup is:
```bash
$ autoreconf --version
autoreconf (GNU Autoconf) 2.69
...

$ make --version
GNU Make 4.1
...

$ doxygen --version
1.8.11

$ pdflatex --version
pdfTeX 3.14159265-2.6-1.40.16 (TeX Live 2015/Debian)
kpathsea version 6.2.1

$ dpkg -s texlive-latex-base
Package: texlive-latex-base
Status: install ok installed
Priority: optional
Section: tex
Installed-Size: 8194
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Multi-Arch: foreign
Source: texlive-base
Version: 2015.20160320-1ubuntu0.1
```

I'll download arch virtual machine image try to reproduce your problem

battlesnake, 2018-01-10:
Odd.  Perhaps a bug upstream in pdflatex or doxygen.

eazrael, 2018-06-15:
I can build documentation with doxygen, but it's empty. this is independent of the output format.

kopasiak, 2018-08-22:
I finally got some spare time to test this issue on arch machine and with versions:
`
$ autoreconf --version
autoreconf (GNU Autoconf) 2.69
...

$ make --version
GNU Make 4.2.1
...

$ doxygen --version
1.8.14

$ pdflatex --version
pdfTeX 3.14159265-2.6-1.40.19 (TeX Live 2018/Arch Linux)
kpathsea version 6.2.3

$ pacman -Qi texlive-bin
Name            : texlive-bin
Version         : 201.47465-3
`

pabs3, 2021-07-13:
@battlesnake did you manage to get it working in the last years?

battlesnake, 2021-07-13:
I had a slight career change and don't work with hardware any more

closed 2021-07-13 by battlesnake

pabs3, 2021-07-14:
OK. FTR as far as I can tell it works for me too.