linux-usb-gadgets/libusbgx #27: Can't compile out of tree

studiofuga, 2018-03-12:
With the current `0.2.0` (999a6e5ee94e572da0ca461099c5006dc65294c7) the library can't be compiled out of the tree:

```
$ autoreconf -i
$ mkdir build
$ cd build
$ ../configure --without-libconfig
$ make -j8
...
libtool: compile:  gcc -DPACKAGE_NAME=\"libusbgx\" -DPACKAGE_TARNAME=\"libusbgx\" -DPACKAGE_VERSION=\"0.2.0\" "-DPACKAGE_STRING=\"libusbgx 0.2.0\"" -DPACKAGE_BUGREPORT=\"k.opasiak@samsung.com\" -DPACKAGE_URL=\"\" -DPACKAGE=\"libusbgx\" -DVERSION=\"0.2.0\" "-D_GNU_SOURCE=/**/" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -I. -I../../src -I../../include/ -g -O2 -MT function/libusbgx_la-midi.lo -MD -MP -MF function/.deps/libusbgx_la-midi.Tpo -c ../../src/function/midi.c  -fPIC -DPIC -o function/.libs/libusbgx_la-midi.o
In file included from ../../src/usbg_common.c:13:0:
../../include/usbg/usbg.h:33:26: fatal error: usbg_version.h: File o directory non esistente
compilation terminated.
make[1]: *** [libusbgx_la-usbg_common.lo] Error 1
make[1]: *** Attesa per i processi non terminati....
In file included from ../../src/usbg_schemes_none.c:13:0:
../../include/usbg/usbg.h:33:26: fatal error: usbg_version.h: File o directory non esistente
compilation terminated.
```

It compiled correctly on 904b04c4b46ec673197ad1d8291a0aabe6a34f52.

WORKAROUND:

Add the proper include path from the build directory when compiling:

```
$ ../configure --without-libconfig CFLAGS="-I${PWD}/include/usbg"
```

studiofuga, 2018-03-12:
This patch fixes it.
[0001-Fix-issue-27-Can-t-compile-out-of-the-tree.patch.txt](https://github.com/libusbgx/libusbgx/files/1803180/0001-Fix-issue-27-Can-t-compile-out-of-the-tree.patch.txt)

kopasiak, 2018-08-22:
Merged. Thank you very much for the patch and sorry for the delay.

closed 2018-08-22 by kopasiak