linux-usb-gadgets/libusbgx #31: Adding FindUsbgx.cmake script with exports as CMake support file

studiofuga, 2018-10-12:
The attached file is useful for use with CMake, since it checks for the library and exports the proper module `Usbgx::Usbgx`.

It should be installed in `${LIBDIR}/cmake/Usbgx`.
Hope it helps.

[FindUsbgx.cmake.txt](https://github.com/libusbgx/libusbgx/files/2473626/FindUsbgx.cmake.txt)

kopasiak, 2018-10-12:
Nice one! How about adding this to repository and creating a suitable makefile target?

studiofuga, 2018-10-12:
I'm not very familiar with autotools. But I'll try. Thanks.

studiofuga, 2018-10-12:
Should I add an additional config option and a conditional test? Perhaps it can always be installed as for the pkgconfig file.

kopasiak, 2018-10-12:
Well... yeah let's install it always. If there is anyone who is annoyed with that he will submit a patch;)

studiofuga, 2018-10-12:
Well, I almost did it but I'm unable to fix the following issue:

I created a proper `LibUsbgxConfig.in` file that should generate `LibUsbgxConfig.cmake` and it makes use of `@libdir@` and `@includedir@` variables, but they expands to `${prefix}/include` and this cannot work. I need fully expanded variables.

Do you know how to do this?
Thanks

kopasiak, 2018-10-12:
@pszewczyk you have some nice experience with autotools. Do you have any idea where the problem comes from?

kopasiak, 2018-10-12:
BTW to simplify the debugging you can push the WIP code as pull request

studiofuga, 2018-10-12:
Ok I'll do

pabs3, 2021-07-13:
@studiofuga did you get the chance to reformat this as a pull request?

studiofuga, 2021-07-13:
hi @pabs3 , my changes were already included in the project, see commit 664b7e6fe7f8269a774101d7625dd0a82c8aadd3 .
I'm not using the library in the project anymore so I can't tell if they work or not.

pabs3, 2021-07-14:
@studiofuga I see, thanks for the info. I'm going to close this then, presumably it works and if it doesn't then someone will likely file a new issue about it.

closed 2021-07-14 by pabs3