linux-usb-gadgets/libusbgx #34: Gadget Printer Support?

ejlevin1, 2019-02-21:
The current function enumeration within libusbgx doesn't support a printer function.  Is this on purpose, or for some reason, or because it wouldn't work for some reason?

kopasiak, 2019-02-21:
The only reason that there is no printer is just because no one added support for it;)
but you are welcomed to do so:)

AristoChen, 2021-02-16:
Hi @kopasiak , 

Sorry for bothering, is this repo still maintained? I found the latest commit is about one year ago, and seems that there is no response for recent open issue and pull request.

If this repo is still under maintained, I am able to help with adding support for this, thanks!

kopasiak, 2021-02-16:
Hi @AristoChen,

to be honest I have way less time recently that I'd love to spend on this.
Any help is very appreciated! You an easily start with reviewing & testing pending pull requests

mgrzeschik, 2021-05-04:
> Hi @AristoChen,
> 
> to be honest I have way less time recently that I'd love to spend on this.
> Any help is very appreciated! You an easily start with reviewing & testing pending pull requests

Hi @kopasiak,

what do you think of moving the project to the new group "linux-usb-gadgets" so it could
be maintained from the whole group?

pabs3, 2021-07-13:
The project moved to @linux-usb-gadgets and the printer function PR #53 by @AristoChen has now been merged into master, please try it and file new PRs (or issues) for any fixes that need to be made.

closed 2021-07-13 by pabs3