linux-usb-gadgets/libusbgx #36: Running Examples

icordova07, 2019-03-21:
After passing the second step outlined in the README file:

`gadget-acm-ecm`

 the following error is presented:

```
usbg_init()  No such file or directory: couldn't init gadget state

Error on USB gadget init
Error: USBG_ERROR_NOT_FOUND : Not found (file or directory removed)

```

How can this issue be resolved?

Also, as a side note, the line 

`make doxygen-doc`

gives out the following error:

`make: *** No rule to make target 'doxygen-doc'.  Stop.`

kopasiak, 2019-04-25:
You probably don't have ConfigFS mounted and/or gadget subsystem loaded

nmay231, 2020-05-31:
I'm not a very advanced user when it comes to this stuff, but this resolved my issue.

```
sudo modprobe g_ether configfs
```

pabs3, 2021-07-13:
@icordova07 I've documented that the configfs Linux kernel module needs to be loaded, please let us know if that works or not for you.

closed 2021-07-13 by pabs3