linux-usb-gadgets/libusbgx #38: Error while running examples on beaglebone black latest image

vaishnav98, 2019-05-21:
I get the following error when i try to run the example:

debian@beaglebone:~/libusbg/examples$ gadget-acm-ecm
usbg_init_state()  Success: unable to parse /sys/kernel/config/usb_gadget

usbg_init()  Success: couldn't init gadget state

Error on USB gadget init
Error: USBG_ERROR_NOT_SUPPORTED : Function not supported

Additional Details:

debian@beaglebone:~/libusbg/examples$ uname -a
Linux beaglebone 4.14.71-ti-r80 #1 SMP PREEMPT Fri Oct 5 23:50:11 UTC 2018 armv7l GNU/Linux
debian@beaglebone:~/libusbg/examples$ cat /etc/dogtag
BeagleBoard.org Debian Image 2018-10-07

However, when I perform the same procedure on older images I can run it without any issues? Can you please help me fix the issue

kopasiak, 2019-05-21:
Could you please try to strace gadget-acm-ecm?

vaishnav98, 2019-05-22:
Thank you for the quick reply, a fresh installation of https://github.com/libusbgx/libusbgx has solved the problem , while reporting the issue I had installed both libusbg and libusbgx and was facing with the errors. uninstalling the libusbg has solved the problem for me.

closed 2019-05-22 by vaishnav98