linux-usb-gadgets/libusbgx #39: Compilation fails on gcc v8

studiofuga, 2019-05-23:
Compilation fails on gcc 8:

```
happycactus@linux-lngr:~/devel/libusbgx> gcc --version
gcc (SUSE Linux) 8.3.1 20190226 [gcc-8-branch revision 269204]
```
```

/usr/lib64/gcc/x86_64-suse-linux/8/../../../../x86_64-suse-linux/bin/ld: /home/happycactus/devel/libusbgx/src/.libs/libusbgx.so: undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:494: gadget-import] Error 1
libtool: link: gcc -g -O2 -o .libs/show-udcs show-udcs.o  -L../src/ /home/happycactus/devel/libusbgx/src/.libs/libusbgx.so
/usr/lib64/gcc/x86_64-suse-linux/8/../../../../x86_64-suse-linux/bin/ld: /home/happycactus/devel/libusbgx/src/.libs/libusbgx.so: undefined reference to `makedev'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:502: gadget-ms] Error 1
/usr/lib64/gcc/x86_64-suse-linux/8/../../../../x86_64-suse-linux/bin/ld: /home/happycactus/devel/libusbgx/src/.libs/libusbgx.so: undefined reference to `makedev'

```

MerlijnWajer, 2020-01-24:
I believe this will fix your problem https://github.com/maemo-leste/libusbgx/commit/2d188e49503c07b74ff029b4c44098849c917971

MerlijnWajer, 2020-01-24:
Would be good to get an upstream approved fix.

commit b5ad66eb (2020-01-30): Fix #39 Compilation fails on gcc v8
closed 2020-01-30 by kopasiak