linux-usb-gadgets/libusbgx #42: Permissions?

sn00pster, 2019-08-10:
You have to run the creation examples as root, any chance of instructions on how to configure the system so that it can be run as a user?

kopasiak, 2019-08-12:
Currently it's not possible. It's not the limitation of the library itself but rather kind of broken ownership management in ConfigFS.

pabs3, 2021-07-13:
Probably the right solution to non-root configuration is a service implementing a dbus interface. There is one called gadgetd available, but it needs a lot of work. That may happen if gadgetd moves to the new @linux-usb-gadgets organisation.

https://github.com/gadgetd/gadgetd

I don't think there is anything to do in libusbgx, so closing this issue.

closed 2021-07-13 by pabs3