linux-usb-gadgets/libusbgx #43: UDC convenience function

sn00pster, 2019-08-14:
You can retrieve the name of the UDC using usbg_get_udc_name which in my case returns "UDC", the content of this file (which you check in some functions) is the real name of the UDC the gadget is attached to, something like 20980000.usb).

A convenience function to get the full "resolved" path to the UDC (i.e the /sys/class/udc/<udc real name>") would be super useful.

p = usbg_get_udc_path(g);

p = "/sys/class/udc/20980000.usb"

I can then use this easily to add a watch on the state file to monitor for the gadget being enumerated and disconnected from a host.

kopasiak, 2019-08-16:
Instead of watch you should rather use udev rules or monitor.

You can check some nice blog post by @andrzejtp how to get this nicely automated with systemd:
https://www.collabora.com/news-and-blog/blog/2019/02/18/modern-usb-gadget-on-linux-and-how-to-integrate-it-with-systemd-part-1/

sn00pster, 2019-08-16:
> https://www.collabora.com/news-and-blog/blog/2019/02/18/modern-usb-gadget-on-linux-and-how-to-integrate-it-with-systemd-part-1/

I can't see anything there relevant to this?  Have I missed something obvious?

Linux gadgets (or rather their UDC's) don't send udev events on being connected to a host or disconnected from a host, this is my fundamental issue.

kopasiak, 2019-08-16:
Yeah, they just send events when they appear in the system.

The functionality that you are looking for is provided by extcon API so you should subscribe to the extcon events

sn00pster, 2019-08-16:
there’s no references in the dwc2 driver (as used on the bcm2385 based raspberry pi devices) to the extcon API, I also believe the USB PHY is built into the DWC2 IP,  so as far as i can tell theres no phy support for extcon either.

I’m not sure if I missed something or it’s just not supported.  Theres absolutely no events whatsoever on the system that I can see when a gadget is connected or disconnected from a host.

kopasiak, 2019-08-16:
Probably you should check your board schematics and look for sth like VBUS detection. On my testing platform (Odroid U3+) I use extcon-gpio driver as it has a separate GPIO line for sensing VBUS

RoganDawes, 2020-03-03:
Is there no way to detect/notify when the USB protocol negotiation happens?

For example, my device could be plugged into a phone charger which would trigger VBUS, but would not mean that there was any actual USB protocol negotiation happening.