linux-usb-gadgets/libusbgx #44: Max power is limited to uint8_t

jackhumbert, 2019-11-04:
In this struct:

```c
struct usbg_config_attrs
{
	uint8_t bmAttributes;
	uint8_t bMaxPower;
};
```

the max power is limited to the uint8_t type, but the system expects the power to be set in mA, not the 2mA units that the USB spec describes, so amounts like `500` get capped to `244`. Would it be appropriate to change the type in this struct, or bit shift the value provided here before passing it along to gadget? [Composite specifies MaxPower as u16](https://www.kernel.org/doc/html/v4.19/driver-api/usb/gadget.html?highlight=composite#c.usb_configuration).

kopasiak, 2019-11-04:
Thank you for catching this.

Taken into account that we expose this structure publicly and that the name is comes straight from the spec (bMaxPower) I'd prefer to keep the same meaning as has been defined in USB spec.

Having said that I'd rather prefer to keep the meaning of this field as defined in USB spec, improve the documentation and multiply this value by 2 while writing it to the kernel.

pabs3, 2021-07-13:
@jackhumbert would you like to prepare a pull request for the method @kopasiak suggests?