linux-usb-gadgets/libusbgx #45: libusbgx uses 0x409 for language instead of 0x0409

buzmeg, 2020-01-09:
This caused me no end of grief today.  I was kept getting USBG_ERROR_NOT_FOUND on my language directories.  It turns out that libusbg regards 0x409 and 0x0409 as different.

It should probably deal with both, but I'm pretty sure that 0x0409 is canonical with the standard, so libusbgx probably needs to accept and emit that.

Thanks.

```
# git diff usbg.c 
diff --git a/src/usbg.c b/src/usbg.c
index d2bf381..86f9820 100644
--- a/src/usbg.c
+++ b/src/usbg.c
@@ -540,7 +540,7 @@ static int usbg_parse_config_strs(const char *path, const char *name,
        int nmb;
        char spath[USBG_MAX_PATH_LENGTH];
 
-       nmb = snprintf(spath, sizeof(spath), "%s/%s/%s/0x%x", path, name,
+       nmb = snprintf(spath, sizeof(spath), "%s/%s/%s/0x%04x", path, name,
                        STRINGS_DIR, lang);
        if (nmb >= sizeof(spath)) {
                ret = USBG_ERROR_PATH_TOO_LONG;
@@ -856,7 +856,7 @@ static int usbg_parse_gadget_strs(const char *path, const char *name, int lang,
        DIR *dir;
        char spath[USBG_MAX_PATH_LENGTH];
 
-       nmb = snprintf(spath, sizeof(spath), "%s/%s/%s/0x%x", path, name,
+       nmb = snprintf(spath, sizeof(spath), "%s/%s/%s/0x%04x", path, name,
                        STRINGS_DIR, lang);
        if (nmb >= sizeof(spath)) {
                ret = USBG_ERROR_PATH_TOO_LONG;

```

kopasiak, 2020-01-30:
Hmmm... What is the kernel reaction if you try to provide both 0x0409 and 0x409?

pabs3, 2021-07-13:
@buzmeg can you check the question that was asked here? Also can you resubmit this patch as a pull request instead, so you can be credited in the commit log?

buzmeg, 2021-07-20:
It's been a while since I have even touched anything USB.  I eventually just decided that I was banging into too many obscure corners of the USB Gadget system and that libusbgx was not really going to be able to help.  Consequently, I haven't looked at this in forever.

Sorry.

buzmeg, 2023-11-20:
Old bug that I no longer can help track.  Closing.