linux-usb-gadgets/libusbgx #47: Create a new GitHub release

rzampino-xilinx, 2020-04-17:
Hello,

Would it be possible to create a new release of libusbgx? The last release (0.2.0) on GitHub is from March 2018 and is missing some changes including the gcc v8 fixes. 
If these releases exist somewhere else that you could point me at that would also be great!

Thank you!

pabs3, 2021-07-13:
We have created a new @linux-usb-gadgets GitHub org and moved libusbgx, gt and ptp-gadget there. We are in the process of merging pull requests and triaging issues. We welcome any help you can give to the project. I expect we will make a release in a while, in the meantime it would be great if you could test git master and file pull requests (or issues) for anything you notice that needs fixing.

pabs3, 2021-07-13:
@rzampino-xilinx see the comment above.

manut, 2022-08-07:
Hi @pabs3 ,
I try to package libusbgx for debian:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1016019
It would be nice to have a new libusbgx release to have the latest fixes in the package.

pabs3, 2022-08-08:
@mgrzeschik is there anything you want to get included before the release? Perhaps your #64 pull request?

@manut same question for you.

manut, 2022-08-08:
@pabs3 I do not have anything to add for the release.

mgrzeschik, 2022-09-08:
I have just merged #64. From my side there is nothing left.

pabs3, 2022-09-09:
Thanks. I'll go through the remaining issues/PRs and forks to see if
there is any other work that should be done before the release.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

pabs3, 2022-12-07:
There is a PR in #68 that looks like it needs to be merged before the
release happens, but it likely breaks ABI. Waiting on the author to
check if that is the case before it can be merged.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

folkertvanheusden, 2023-12-08:
@pabs3 I saw that the #68 issue is closed. can you please consider adding a release tag? The uvc-functionality would be great to have! (in e.g. the debian package cc @manut )

pabs3, 2023-12-09:
That pull request wasn't merged, instead the proposer didn't fix the
requested changes, and deleted their branch and closed the request.

I'm not sure when I'll have time for a release, if one of the other
project members wants to do it, please go ahead. That said I did
notice recently that there are at least some compiler and or
analysis tools warnings that should be mopped up first.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

folkertvanheusden, 2023-12-09:
@pabs3 please consider #84 regarding the compile warnings.

mgrzeschik, 2024-09-24:
I plan to release the package today. Including the meson and 9pfs mr. Any more suggestions?

folkertvanheusden, 2024-09-24:
please go ahead :-)

mgrzeschik, 2024-10-16:
For now I skipped the #68 which was closed but I reopened with #100. Since the ABI change is rather big and I for now wanted a release early we can add this with e.g. 0.4.0.