linux-usb-gadgets/libusbgx #49: WinUSB bulk in/out device

rundekugel, 2020-09-18:
I created an small embedded WinUSB device which offers 2 bulk endpoints. This device can communicate with Linux and with Windows10 without installing driver, or a .inf file. And it's not recognized as serial com port.

Now I try to build an embedded Linux gadget with your libusbgx. It must respond to OS_DESC (0xee) as WinUSB device. And must not be recognized as serial com port. It shall behave like a _g_serial_ with param "use_acm=no"

If I build it with filesystem commands. Then the os_desc setting in Window10 it's recognized in registry settings

> HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\usbflags\<VIDPID> 
> osvc = 0xbc

But not recognized as WinUSB device

It looks like, the os_desc feature does only fully work for RNDIS. 

This is the respond to vendor request, which works when I send it from my embedded WinUSB to Windows:
 
> const U8 u8ExtendedCompatIDOSFeatDesc[] = 
>   {
>       0x28, 0x00, 0x00, 0x00, /* dwLength Length of this descriptor */ //(40 bytes)
>       0x00, 0x01,             /* bcdVersion = Version 1.0 */ 
>       0x04, 0x00,             /* wIndex = 0x0004 */ 
>       0x01,                   /* bCount = 1 */ 
>       0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, /* Reserved */ 
>       0x00,                   /* Interface number = 0 */ 
>       0x01,                   /* Reserved */ 
>       0x57, 0x49, 0x4E, 0x55, 0x53, 0x42, 0x00, 0x00, //string = "WINUSB"
>       0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, /* subCompatibleID */ 
>       0x00, 0x00, 0x00, 0x00, 0x00, 0x00  /* Reserved */   
>   };

I tried with a modified example of your code: 
[myWinusbGadget.zip](https://github.com/libusbgx/libusbgx/files/5245446/myWinusbGadget.zip)

How can I make it working for _gser_ also? Is it possible with changing your src? Or do I have to change the kernel gadget files?
Can you help me to add this feature to your lib?

pabs3, 2021-07-13:
@rundekugel are you able to trace the `usbg_set_interf_os_desc` in a debugger to find out where the `USBG_ERROR_NOT_FOUND` error is coming from? The function itself does not set that error, but other functions might do that.

rundekugel, 2021-09-17:
@pabs3 sorry, I found no time to trace the issue. 
I used https://github.com/rundekugel/gadgetfsd/tree/WinUSB to get my project running.

pabs3, 2021-09-18:
Please let us know if you find time to trace the issue.

I think you have provided enough info about [WinUSB](https://en.wikipedia.org/wiki/WinUSB), thanks.

For the sake of completeness, which versions of the Linux kernel are you finding the error on? Have you tried the most recent ones (5.15-rc1 right now or 5.14.5 stable)?

Unfortunately I don't have any physical USB UDC gadget devices nor do I have a Windows install, so I won't be able to debug this either.

rundekugel, 2021-12-13:
I'm sorry, I don't find time to trace the issue.