linux-usb-gadgets/libusbgx #58: gt enable <gadget> returns error string: Failed to enable gadget Invalid parameter

Dvergatal, 2021-09-30:
Just like in the subject the gt app returns error string _Failed to enable gadget Invalid parameter_. The error code **USBG_ERROR_INVALID_PARAM** is being returned from method _usbg_enable_gadget_ in libusbgx/src/usbg.c. I have verified that it is being returned on no udc. I suspect that i am missing some kernel configuration for UDC, because even /sys/class/udc is empty. Correct me if i'm wrong, because i have no other clues.

pabs3, 2021-09-30:
Looking at the code, your analysis sounds correct.

The usbg_parse_udcs function fills in the udcs list from /sys/class/udc
and if there are none then usbg_get_first_udc returns NULL and so
usbg_enable_gadget will exit with USBG_ERROR_INVALID_PARAM.

If /sys/class/udc is empty then your Linux kernel version either
doesn't have any UDC support compiled in (grep for UDC in your Linux
kernel build config), or your device doesn't have any UDC hardware (you
can use the dummy UDC to workaround that).

https://www.collabora.com/news-and-blog/blog/2019/06/24/using-dummy-hcd/

Which device do you have?

Which UDC hardware does your device have?

Which version of the Linux kernel do you use?

Which Linux kernel build config are you using?

Which version of libusbgx/gt are you using?

If you build (with CONFIG_USB_DUMMY_HCD=m) and load the dummy UDC
(modprobe dummy_hcd), does libusbgx and gt work?

Arguably usbg_enable_gadget should return a value other than
USBG_ERROR_INVALID_PARAM for situations where no UDC is available,
would you be interested in helping contribute a patch for this?

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

Dvergatal, 2021-09-30:
Hi @pabs3,
i'm using firefly-rk3399 and with old 4.4.194 rockchip kernel, it was working alright. Now i have switched to 5.4.18 because of WPA3 support in it. I can debug which udc it was by switching to old kernel image. and also check your suggestion. Btw. yes i can help in cotribute a patch for this.

Dvergatal, 2021-09-30:
Ok i have found out that with an older kernel i have this fe800000.dwc3 -> ../../devices/platform/usb0/fe800000.dwc3/udc/fe800000.dwc3 in older kernel but i have dwc3 support added in kernel 5.4.18 so i don't know what i can be missing.

Dvergatal, 2021-10-01:
BTW. sorry for not answering first your questions.

> Which device do you have?
firefly-rk3399

> Which UDC hardware does your device have?
fe800000.dwc3

> Which version of the Linux kernel do you use?
5.4.18

> Which Linux kernel build config are you using?
defconfig for arm64

> Which version of libusbgx/gt are you using?
libusbgx-0.2.0
gt-afe7d8550ad1f0d6c97deefd645824b98397e33e from andrzejtp github

> If you build (with CONFIG_USB_DUMMY_HCD=m) and load the dummy UDC
(modprobe dummy_hcd), does libusbgx and gt work?
Not tried it yet.

And i have been talking yesterday on irc with @mmind and he told me that there is a driver issue. On vendor kernel which is 4.4.194 rockchip team hacked that extcon thingy into the fusb302 driver, but that is of course no viable solution for mainline. For 5.4.18 i need to write a patch in which i will use the type-c framework in phy_rockchip_typec driver wich will know to use typec_fusb302 to make a call to dwc3. So i think we can close this issue.

pabs3, 2021-10-01:
Thanks for tracking the issue down, good luck with the driver fixes, closing this issue.

@Dvergatal If you are planning on writing a patch clarifying the libusbgx/gt error when no UDC is available, that would be great. Otherwise please file a separate issue for that and someone else will take care of it at some point.

closed 2021-10-01 by Dvergatal