linux-usb-gadgets/libusbgx #65: Improve net function to use Windows 10 native RNDIS driver

fginez, 2022-09-12:
I am developing a multifunction gadget device with 2 different interfaces: HID and RNDIS.

The HID interface is working properly, but I coudn't make the RNDIS interface work with Windows 10 native RNDIS driver without manually writting some specific class/subclass/protocol values to the function attributes using the 'echo' command in the sysfs's exposed files. I coudn't manage to find another way to make it work when using a multifunction gadget.

By default, class, subclass and protocol attributes (inside rndis function folder) have the following values: 02 (CDC), 06 (NCM) and 00 (No specific protocol), and as far as I know it isn't possible to change these values using libusbgx.

I wrote a patch (attached) which allows the user to write any value to the above mentioned attributes, and specifically for RNDIS usage the values are: EF (Miscellaneous), 04 (RNDIS) and 01 (RNDIS over Ethernet).

I'd like to help another developers which may face a similar issue with the W10 driver. Do you think it worths to merge this feature to libusbgx?

[net_attr.txt](https://github.com/linux-usb-gadgets/libusbgx/files/9551073/net_attr.txt)

pabs3, 2022-09-19:
@fginez thanks for the contribution!

This seems like a reasonable feature to add, could you please re-submit
it as a pull request so that the appropriate git commit metadata will
be present in the libusbgx git history after merging the change?

If you want to add support for the feature in gt/GadgetTool that would
be nice too, so that it is available from the command-line as well.

https://github.com/linux-usb-gadgets/gt/

PS: I am interested to hear more about the situation where your
combined HID/RNDIS gadget will be used if you are willing to share.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

moether, 2022-10-31:
@fginez It seems you have committed your changes to master.
However, the following line clashes when the header is included in a c++ project as you use the keyword "class"

https://github.com/linux-usb-gadgets/libusbgx/blob/master/include/usbg/function/net.h#L32

pabs3, 2022-10-31:
Does wrapping it in `extern "C" {}` not mitigate this issue?

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

moether, 2022-10-31:
It's not. Initially, I also thought it does.
Here is a simple example: [https://onlinegdb.com/paMKBfwo6](https://onlinegdb.com/paMKBfwo6)
Just by renaming the struct member from "class" to anything else makes it compile again

pabs3, 2022-11-01:
OK, thanks. I've renamed the variables to class_ instead:

https://github.com/linux-usb-gadgets/libusbgx/commit/36e71e1e1ea9fb18d4765d0376329dcde90c8c91

I'll close this issue now since the patch is merged and fixed.
Please open new issues or pull requests for other matters.

While I have you, I'm interested to hear about the context you are
using libusbgx in if you are willing to share those details.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

closed 2022-11-01 by pabs3

moether, 2022-11-01:
We use it in an embedded Linux project to programmatically configure the usb gadget modes from c++ software.
Doing all that just by manually creating all those directories and files in the configfs is quite a bit pain in the ***. We find it way more comfortable to use this library.
Thanks for maintaining it btw!

pabs3, 2022-11-01:
Thanks for the info!

-- 
bye,
pabs

https://bonedaddy.net/pabs3/