linux-usb-gadgets/libusbgx #75: uac c_srate/c_srare attribute suport for multiple samplerates

bitkeeper, 2023-04-16:
Since the 5.18+ kernel UAC supportes multiple samplerates. See [u_audio.h](https://elixir.bootlin.com/linux/latest/source/drivers/usb/gadget/function/u_audio.h#L41).

Would it be possible to support this?

pabs3, 2023-04-17:
That sounds like a useful feature that we would accept patches for.

Unfortunately the project doesn't currently have people working on
feature requests and there aren't any people funded to work on the
project, so it could be some time until this is implemented, unless
you are interested in working on this yourself.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

bitkeeper, 2023-04-19:
ok fair. 

I couldn't directly found an example of another scheme that is using arrays with regular values.
Can you at least point me in the correct direction where/how to this.

So far I found:
* uac2.h : modify the struct members to hold the same fields as the kernel driver (change the rates to arrays):
```c
struct usbg_f_uac2_attrs {
	int c_chmask;
	int c_srate[UAC_MAX_RATES]; # < modified
	int c_ssize;
	int p_chmask;
	int p_srate[UAC_MAX_RATES]; # < modified
	int p_ssize;
};

union usbg_f_uac2_attr_val {
	int c_chmask;
	int c_srate[UAC_MAX_RATES]; # < modified
	int c_ssize;
	int p_chmask;
	int p_srate[UAC_MAX_RATES]; # < modified
	int p_ssize;
};
```
Or is better to use a dynamic list here?

* next is to uac2.c register different set/get/import/export which are name based
```
#define UAC2_DEC_ATTR(_name)						\
	{								\
		.name = #_name,						\
		.offset = offsetof(struct usbg_f_uac2_attrs, _name),    \
		.get = usbg_get_dec,				        \
		.set = usbg_set_dec,				        \
		.import = usbg_get_config_node_int,	                \
		.export = usbg_set_config_node_int,		        \
	}

#define UAC2_DEC_ARRAY_ATTR(_name)						\
	{								\
		.name = #_name,						\
		.offset = offsetof(struct usbg_f_uac2_attrs, _name),    \
		.get = usbg_get_dec_array,				        \
		.set = usbg_set_dec_array,				        \
		.import = usbg_get_config_node_int_array,	                \
		.export = usbg_set_config_node_int_array,		        \
	}

static struct {
	const char *name;
	size_t offset;
	usbg_attr_get_func get;
	usbg_attr_set_func set;
	usbg_import_node_func import;
	usbg_export_node_func export;
} uac2_attr[USBG_F_UAC2_ATTR_MAX] = {
	[USBG_F_UAC2_C_CHMASK] = UAC2_DEC_ATTR(c_chmask),
	[USBG_F_UAC2_C_SRATE] = UAC2_DEC_ARRAY_ATTR(c_srate),
	[USBG_F_UAC2_C_SSIZE] = UAC2_DEC_ATTR(c_ssize),
	[USBG_F_UAC2_P_CHMASK] = UAC2_DEC_ATTR(p_chmask),
	[USBG_F_UAC2_P_SRATE] = UAC2_DEC_ARRAY_ATTR(p_srate),
	[USBG_F_UAC2_P_SSIZE] = UAC2_DEC_ATTR(p_ssize),
};
```

* And now implement the array functions:
  * usbg_get_dec_array
  * usbg_set_dec_array
  * usbg_get_config_node_int_array  - configlib importer
  * usbg_set_config_node_int_array  - configlib exporter

What else should be done?

Thanks!

pabs3, 2023-04-20:
I think this needs a bit more careful implementation because:

The library should build and work on older Linux kernel versions
and preferably work on newer versions when built on older versions
and vice versa, if those are possible.

The library should avoid changing ABI so that applications using 
it do not need to be recompiled in order to continue working.

So, directly changing structures/functions should be avoided
and instead new structures/functions should be added.

Maybe the correct Linux kernel API to use should be auto-chosen at
runtime, or maybe the libusbgx functions should pass through failure
and expect the application using it to handle different versions?

Other than that, and my relative lack of experience with this library,
your proposed changes sound like the right way to go.

PS: since your GitHub account is a shared one, please ensure that your
commits correctly attribute the person doing the code changes, rather
than attributing the shared account. If there are multiple people
working on this, you can attribute them with Co-authored-by footers.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/