linux-usb-gadgets/libusbgx #76: doesn't work with kernel 6.1.34+

folkertvanheusden, 2023-06-25:
gadget-uvc fails with:

```
symlink("/sys/kernel/config/usb_gadget/g1/functions/uvc.uvc", "/sys/kernel/config/usb_gadget/g1/configs/The only one.1/uvc.cam") = 0
openat(AT_FDCWD, "/sys/kernel/config/usb_gadget/g1/UDC", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=0, ...}) = 0
write(3, "20980000.usb", 12)            = -1 EBUSY (Device or resource busy)
close(3)                                = 0
write(2, "Error enabling gadget\n", 22Error enabling gadget
) = 22
write(2, "Error: USBG_ERROR_BUSY : Busy (g"..., 47Error: USBG_ERROR_BUSY : Busy (gadget enabled)
) = 47
exit_group(-22)                         = ?
```

and in dmesg it says:

```Sun Jun 25 20:54:25 2023] UDC core: g1: couldn't find an available UDC or it's busy```

pabs3, 2023-06-26:
Does it work with other gadgets?

Does it work with older/newer Linux kernel versions?

Which hardware is this? Does it work with other boards?

Which UDC is this? Does it work with other ones or the virtual one?

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

folkertvanheusden, 2023-06-26:
Hi,

It works fine with the ethernet gadget.

I've tried several versions upto 4.19.118 and none worked.

This is the Raspberry Pi Zero W / WH. I don't have other boards to test with.

I don't know about the UDC. What other ones?

folkertvanheusden, 2023-06-27:
Found the problem: altough in my google bubble every page says that I need to modprobe g_webcam, it should be usb_f_uvc.

closed 2023-06-27 by folkertvanheusden

pabs3, 2023-06-28:
Usually boards only have one UDC, but there is also a virtual UDC:

https://www.collabora.com/news-and-blog/blog/2019/06/24/using-dummy-hcd/

I wonder if we should document somewhere that usb_f_uvc should be used
instead of g_webcam.

I also wonder if g_webcam should get fixed so it works like usb_f_uvc.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/