linux-usb-gadgets/libusbgx #77: how to find the gadget device file?

folkertvanheusden, 2023-06-27:
Hi,

When I create an UVC device, a /dev/video1 appears. How do I find that device file programmatically? I know that udev can signal this, but how do I then know that the newly created device is the one I just initialized and not some other by an other application?

pabs3, 2023-06-28:
I don't know enough Linux USB gadget stuff to answer this properly,
but in the worst case you can always filter new devices to just the
ones that match exactly the one that you configured, inc serial num?

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

folkertvanheusden, 2023-06-28:
I was pointed at https://gitlab.freedesktop.org/camera/uvc-gadget/-/blob/master/lib/configfs.c#L199 so I'm going for that route.

closed 2023-06-28 by folkertvanheusden