linux-usb-gadgets/libusbgx #78: Printer function header not installed

nl250060, 2023-07-28:
The header file for the USB printer function does not get installed

nl250060, 2023-07-28:
The following patch fixes this for me.

```
From 19cdfd4540136aab0040ca448c40195ff1c5713c Mon Sep 17 00:00:00 2001
From: Niall Leonard <nl250060@ncr.com>
Date: Fri, 28 Jul 2023 12:41:14 +0100
Subject: [PATCH] Added printer function header to installed headers

---
 Makefile.am | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/Makefile.am b/Makefile.am
index 309c539..c12628b 100644
--- a/Makefile.am
+++ b/Makefile.am
@@ -14,7 +14,7 @@ EXTRA_DIST = doxygen.cfg
 library_includedir=$(includedir)/usbg
 library_include_HEADERS = include/usbg/usbg.h include/usbg/usbg_version.h
 function_includedir=$(includedir)/usbg/function
-function_include_HEADERS = include/usbg/function/ffs.h include/usbg/function/loopback.h include/usbg/function/midi.h include/usbg/function/ms.h include/usbg/function/net.h include/usbg/function/phonet.h include/usbg/function/serial.h include/usbg/function/hid.h include/usbg/function/uac2.h include/usbg/function/uvc.h
+function_include_HEADERS = include/usbg/function/ffs.h include/usbg/function/loopback.h include/usbg/function/midi.h include/usbg/function/ms.h include/usbg/function/net.h include/usbg/function/phonet.h include/usbg/function/serial.h include/usbg/function/hid.h include/usbg/function/uac2.h include/usbg/function/uvc.h include/usbg/function/printer.h
 pkgconfigdir = $(libdir)/pkgconfig
 pkgconfig_DATA = libusbgx.pc
 cmakeconfigdir = $(libdir)/cmake/LibUsbgx
```

pabs3, 2023-07-29:
Merged. Next time, please submit pull requests instead of patches in issues. Please also mention the commit you are fixing in the commit message, I added it manually this time.

closed 2023-07-29 by pabs3