linux-usb-gadgets/libusbgx #86: What about  inquiry_string?

NekoCWD, 2024-02-21:
Mass storage have one more option called "inquiry_string"

It made for setting custom  scsi name for mass storage.

It shown as device name in my BIOS and in dmesg

There are example:
I runed
```
echo "MEOW" > inquiry_string
```
Before it:

```
[timesnap] scsi 6:0:0:0: Direct-Access     Linux    File-Stor Gadget 0607 PQ: 0 ANSI: 2
[timesnap] scsi 6:0:0:1: CD-ROM            Linux    File-Stor Gadget 0607 PQ: 0 ANSI: 2
```
After it:

```
[timesnap] scsi 6:0:0:0: Direct-Access     MEOW                           PQ: 0 ANSI: 2
[timesnap] scsi 6:0:0:1: CD-ROM            MEOW                           PQ: 0 ANSI: 2
```

pabs3, 2024-02-22:
A pull request for this would be welcome, but we don't yet have enough
resources to implement this ourselves.

-- 
bye,
pabs

https://bonedaddy.net/pabs3/

NekoCWD, 2024-02-27:
Merged

closed 2024-02-27 by NekoCWD