linux-usb-gadgets/libusbgx #90: Questionable usage of DEFAULT_UDC in examples

wbehrens-on-gh, 2024-07-12:
DEFAULT_UDC is defined to NULL here
https://github.com/linux-usb-gadgets/libusbgx/blob/a5bfa81017a9b2064bc449cf74f5f9d106445f62/include/usbg/usbg.h#L50

and is used in all the examples when calling `usbg_enable_gadget()`. This function will perform a NULL check on the udc though causing it to always fail when using DEFAULT_UDC (like in the examples

https://github.com/linux-usb-gadgets/libusbgx/blob/a5bfa81017a9b2064bc449cf74f5f9d106445f62/src/usbg.c#L2553

wbehrens-on-gh, 2024-07-15:
Was looking at to much code and missed the obvious line where udc gets set. Why was this reopened?

closed 2024-07-23 by wbehrens-on-gh

pabs3, 2024-08-03:
Sorry, I reopened because I didn't see a reason for the closing for the issue, so I assumed it was still relevant and should be kept option until it gets fixed.