linux-usb-gadgets/libusbgx #102: Error fetching ether gadget attributes

pboucherat-thegoodpenguin, 2024-10-30:
The show-gadgets example fails when trying to display an ether gadget's attributes. The example programs e.g. gadget-acm-ecm create the gadgets passing a null attributes pointer, and the ether gadget only creates 4 attributes dev_addr, host_addr, ifname and qmult, whereas in src/function/ether.c there are 7 attributes defined so  usbg_f_net_get_attrs() fails when  it tries to fetch the fifth one (class_) and returns USBG_ERROR_NOT_FOUND. So after running gadget-acm-ecm, output from show-gadgets looks like this with the not found error where it should be listing the ecm.usb0 attributes : 
```
ID 1d6b:0104 'g1'                                                                                                                                                                                                                                                        
  UDC                   38100000.usb                                                                                                                                                                                                                                     
  bcdUSB                2.01                                                                                                                                                                                                                                             
  bDeviceClass          0x00                                                                                                                                                                                                                                             
  bDeviceSubClass       0x00                                                                                                                                                                                                                                             
  bDeviceProtocol       0x00                                                                                                                                                                                                                                             
  bMaxPacketSize0       64                                                                                                                                                                                                                                               
  idVendor              0x1d6b                                                                                                                                                                                                                                           
  idProduct             0x0104
  bcdDevice             0.01
  Language:     0x409
    Manufacturer        Foo Inc.
    Product             Bar Gadget
    Serial Number       0123456789
  Function, type: acm instance: usb0
    port_num            0
  Function, type: acm instance: usb1
    port_num            1
Error: USBG_ERROR_NOT_FOUND : Not found (file or directory removed)
  Configuration: 'The only one' ID: 1
    MaxPower            2
    bmAttributes        0x80
    Language:   0x409
      configuration     CDC 2xACM+ECM
    acm.GS0 -> acm usb0
    acm.GS1 -> acm usb1
    ecm.usb0 -> ecm usb0

```

pboucherat-thegoodpenguin, 2024-10-30:
I suppose an easy fix would be to remove the class_, subclass and protocol attributes, but I don't know if these are used by anything.

pboucherat-thegoodpenguin, 2024-10-30:
Output looks ok with the attached patch, but I don't know the code well enough to know if this is the best way : 
```
[0001-Fix-get-attibutes-function-for-ether.c.txt](https://github.com/user-attachments/files/17576970/0001-Fix-get-attibutes-function-for-ether.c.txt)

ID 1d6b:0104 'g1'
  UDC                   38100000.usb
  bcdUSB                2.01
  bDeviceClass          0x00
  bDeviceSubClass       0x00
  bDeviceProtocol       0x00
  bMaxPacketSize0       64
  idVendor              0x1d6b
  idProduct             0x0104
  bcdDevice             0.01
  Language:     0x409
    Manufacturer        Foo Inc.
    Product             Bar Gadget
    Serial Number       0123456789
  Function, type: acm instance: usb0
    port_num            0
  Function, type: acm instance: usb1
    port_num            1
  Function, type: ecm instance: usb0
    dev_addr            c6:5d:d:84:9a:ab
    host_addr           7a:16:a2:45:d9:b0
    ifname              usb0
    qmult               5
  Configuration: 'The only one' ID: 1
    MaxPower            2
    bmAttributes        0x80
    Language:   0x409
      configuration     CDC 2xACM+ECM
    acm.GS0 -> acm usb0
    acm.GS1 -> acm usb1
    ecm.usb0 -> ecm usb0

```

elryno, 2024-11-14:
I see the same issue. I am using a vendor kernel based on 5.4.212. Is there a specific kernel version that is required for the latest version of libusbgx?

elryno, 2024-11-14:
Relates to Issue #101

pboucherat-thegoodpenguin, 2024-11-15:
I was also using a 5.15 kernel (can't remember the exact version) on a toradex verdin dev board

einarjon, 2024-12-17:
@elryno, @pboucherat-thegoodpenguin: These class, subclass and protocol variables only exist for RNDIS, but the usbg_f_net_attr(s) struct/enums are also used for [NCM, ACM/ECM, EEM](https://github.com/linux-usb-gadgets/libusbgx/blob/dbedf16f80c5b4e1807ea4f80cc89d1f15d36f0a/src/function/ether.c#L216-L224) and so on that don't have these values  in any kernel.

I posted a different kind of patch in #101 that will make it work for rndis, but this would needs a bit of logic in the getter/setter functions to check if the type is RNDIS, and supply the bigger value in that case.

 I am using both RNDIS and NCM, and for the NCM I just had to revert commits 4ca4a7699 and 36e71e1e1e.